pimoroni/mlx90640-library #8: negative green

timsoftgit, 2019-04-06:
in python/test.py temp_to_col(val) function can return negative values for green (g)
r,g,b = temp_to_col(val)

Gadgetoid, 2019-04-08:
Looks like `colorsys.hsv_to_rgb` does not deal as well with negative input as I had expected. Since the `hue` component of `hsv` can be conceptualised as wrapped around a cylinder I mistakenly assumed that an input of `-0.1` would be identical to `0.9`. Apparently not. Here's a graph that demonstrates just how bonkers it behaves with an negative input (left side of the 0 line) versus positive (right side):

[image]

I suppose I could take any negative `hue` value and `hue = 1.0 - hue` to shift it into the sane, positive range.

Gadgetoid, 2019-04-08:
Hopefully fixed in 191757d517d7e135a1f86a0ac15c89d7b4daed10

closed 2019-04-08 by Gadgetoid