pimoroni/mlx90640-library #10: baudrate variable not used in python driver

timsoftgit, 2019-04-09:
in  mlx90640-python.cpp although baudrate variable is set line 19, on line 65 a hardcoded value is used, and then the following switch statements (73-104) which set baudrate variable depending on fps is done after the actual baudrate is set on line 65.
shouldn't baudrate be set before 65 and make 65 bcm2835_i2c_set_baudrate(baudrate) You would have to split the case statement though, as you can't write using MLX90640_SetRefreshRate until after the bus baudrate has been set on line 65.
you could use an if statement for setting baud rate before 65,
if (fps<16) {baudrate=400000;} else {baudrate=1000000;}
and use the switch statement afterwards (for just setting refresh rate on camera)

Gadgetoid, 2019-04-11:
I'm not convinced that on-the-fly baudrate changes work at all, so this might be something I remove altogether in favour of using `dtparam=i2c1_baudrate=<value>` in `/boot/config.txt`. This is something I need to investigate- it took something of a back seat to getting this sensor working at all.

timsoftgit, 2019-04-11:
sure. It looks like it uses the bcm driver rather than the linux i2c driver anyway. I have my portable heatcam working,( all-be-it very slowly,) which was my main objective. I was looking through the code to remove printf stuff, which is how I noticed.