pimoroni/mlx90640-library #11: No such file or directory  #include <SDL2/SDL.h>

robmarkcole, 2019-04-10:
On running `make` from root, I get the error:

```
pi@bdx92244cee:~/mlx90640/mlx90640-library $ make
g++ -I. -std=c++11 -std=c++11 `sdl2-config --cflags --libs`   -c -o examples/src/sdlscale.o examples/src/sdlscale.cpp
/bin/sh: 1: sdl2-config: not found
examples/src/sdlscale.cpp:10:22: fatal error: SDL2/SDL.h: No such file or directory
 #include <SDL2/SDL.h>
                      ^
compilation terminated.
<builtin>: recipe for target 'examples/src/sdlscale.o' failed
make: *** [examples/src/sdlscale.o] Error 1
```

timsoftgit, 2019-04-10:
sounds like you forgot to install SDL2 development library
`sudo apt install libsdl2-dev`
check REAME.md for all the things you need to install.

robmarkcole, 2019-04-10:
@timsoftgit you are correct! Don't know how I missed that, thanks!

closed 2019-04-10 by robmarkcole