pimoroni/mlx90640-library #12: make I2C_MODE=LINUX fails

robmarkcole, 2019-04-10:
On pi, I get:
```
pi@bdx92244cee:~/mlx90640/mlx90640-library $ make I2C_MODE=LINUX
g++ -I. -std=c++11 -fPIC -I headers -shared    -c -o functions/MLX90640_LINUX_I2C_Driver.o functions/MLX90640_LINUX_I2C_Driver.cpp
In file included from functions/MLX90640_LINUX_I2C_Driver.cpp:24:0:
/usr/include/linux/i2c.h:68:8: error: redefinition of ‘struct i2c_msg’
 struct i2c_msg {
        ^~~~~~~
In file included from functions/MLX90640_LINUX_I2C_Driver.cpp:23:0:
/usr/include/linux/i2c-dev.h:37:8: error: previous definition of ‘struct i2c_msg’
 struct i2c_msg {
        ^~~~~~~
In file included from functions/MLX90640_LINUX_I2C_Driver.cpp:24:0:
/usr/include/linux/i2c.h:131:7: error: redefinition of ‘union i2c_smbus_data’
 union i2c_smbus_data {
       ^~~~~~~~~~~~~~
In file included from functions/MLX90640_LINUX_I2C_Driver.cpp:23:0:
/usr/include/linux/i2c-dev.h:89:7: error: previous definition of ‘union i2c_smbus_data’
 union i2c_smbus_data {
       ^~~~~~~~~~~~~~
functions/MLX90640_LINUX_I2C_Driver.cpp: In function ‘int MLX90640_I2CRead(uint8_t, uint16_t, uint16_t, uint16_t*)’:
functions/MLX90640_LINUX_I2C_Driver.cpp:51:27: error: invalid conversion from ‘__u8* {aka unsigned char*}’ to ‘char*’ [-fpermissive]
     i2c_messages[0].buf = (__u8*)cmd;
                           ^~~~~~~~~~
functions/MLX90640_LINUX_I2C_Driver.cpp:56:27: error: invalid conversion from ‘__u8* {aka unsigned char*}’ to ‘char*’ [-fpermissive]
     i2c_messages[1].buf = (__u8*)buf;
                           ^~~~~~~~~~
functions/MLX90640_LINUX_I2C_Driver.cpp: In function ‘int MLX90640_I2CWrite(uint8_t, uint16_t, uint16_t)’:
functions/MLX90640_LINUX_I2C_Driver.cpp:93:27: error: invalid conversion from ‘__u8* {aka unsigned char*}’ to ‘char*’ [-fpermissive]
     i2c_messages[0].buf = (__u8*)cmd;
                           ^~~~~~~~~~
<builtin>: recipe for target 'functions/MLX90640_LINUX_I2C_Driver.o' failed
make: *** [functions/MLX90640_LINUX_I2C_Driver.o] Error 1
```

timsoftgit, 2019-04-10:
remove libi2c-dev and it works. (it contains duplicate definitions with this lib, which is what gives the error)
see https://forums.pimoroni.com/t/thermal-camera-mlx90640/9802/6
although it would be worth updating the readme.

robmarkcole, 2019-04-10:
Thanks again @timsoftgit yes there are a couple of comments about libi2c-dev [here](https://shop.pimoroni.com/products/mlx90640-thermal-camera-breakout)

Gadgetoid, 2019-04-11:
Thanks for pointing this out. See PR #17 which should, hopefully, make this a non-issue for future users.

closed 2019-04-15 by Gadgetoid