pimoroni/mlx90640-library #13: python/rgb-to-gif.py

robmarkcole, 2019-04-10:
Related to being unable to run `make`, I get:
```
pi@bdx92244cee:~/mlx90640/mlx90640-library $ python python/rgb-to-gif.py 
Traceback (most recent call last):
  File "python/rgb-to-gif.py", line 17, in <module>
    raise RuntimeError("{} doesn't exist, did you forget to run \"make\"?".format(RAW_RGB_PATH))
RuntimeError: ../examples/rawrgb doesn't exist, did you forget to run "make"?
```

timsoftgit, 2019-04-10:
you need to cd to python directory, run 
```
python setup.py build
sudo python setup.py install
```

robmarkcole, 2019-04-10:
HI @timsoftgit I did not realise this required a `build`. Does it matter which dir the command is run from? I am getting:
```
pi@bdx92244cee:~/mlx90640-library/python/library $ python setup.py build
running build
running build_py
running build_ext
building '_MLX90640' extension
creating functions
creating build
creating build/temp.linux-armv7l-2.7
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-Ub3vap/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I../headers -I/usr/include/python2.7 -c ../../functions/MLX90640_API.cpp -o build/temp.linux-armv7l-2.7/../../functions/MLX90640_API.o
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
../../functions/MLX90640_API.cpp:17:33: fatal error: MLX90640_I2C_Driver.h: No such file or directory
 #include <MLX90640_I2C_Driver.h>
                                 ^
compilation terminated.
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
```

robmarkcole, 2019-04-11:
OK I created a completely fresh directory and re-ran the build procedures, this works presumably as I have `apt-get` all the required packages. However now I have https://github.com/pimoroni/mlx90640-library/issues/14 I am able to `sudo examples/rawrgb
`

closed 2019-04-11 by robmarkcole