pimoroni/mlx90640-library #14: Error trying to run rbg-to-gif.py

jep5350, 2019-04-10:
> rgb-to-gif.py - output a gif using ./rawrgb command.
> 
> Use ./rawrgb to grab frames from the MLX90640
> and render them into a gif.
> 
> You must have built the "rawrgb" executable first with "make"
> 
> Press Ctrl+C to save & exit!
> 
> 
> Got 0 bytes of data!
> Traceback (most recent call last):
>   File "rgb-to-gif.py", line 40, in <module>
>     image = Image.frombytes('RGB', (24, 32), frame)
>   File "/usr/lib/python3/dist-packages/PIL/Image.py", line 2100, in frombytes
>     im.frombytes(data, decoder_name, args)
>   File "/usr/lib/python3/dist-packages/PIL/Image.py", line 742, in frombytes
>     raise ValueError("not enough image data")
> ValueError: not enough image data
>

robmarkcole, 2019-04-11:
I think this is the same issue as https://github.com/pimoroni/mlx90640-library/issues/13

robmarkcole, 2019-04-11:
I have this issue, and have verified that ` sudo examples/rawrgb` returns data

closed 2019-04-11 by Gadgetoid