pimoroni/mlx90640-library #16: Gifs and images have checkerboard pattern with rgb-to-gif.py

robmarkcole, 2019-04-11:
As title, example below

[image]

Files [here](https://github.com/robmarkcole/mlx90640-library/tree/adds-rgd-to-file/python)

Gadgetoid, 2019-04-11:
This may be due to your camera drifting out of sync. Here's an example I recorded while putting the python together: https://twitter.com/Gadgetoid/status/1115297867755814914

Have you modified the framerate in your version of rawrgb at all?

robmarkcole, 2019-04-11:
No, is experimentation required?

Gadgetoid, 2019-04-11:
Looks like there's something wrong with my conversion to gif- I think I'm treating the image as being the wrong aspect ratio perhaps.

I'm working on adding an `--fps` param, too, so it's easier to dial it down until it works.

Gadgetoid, 2019-04-11:
This certainly helped fix the problem: https://github.com/pimoroni/mlx90640-library/commit/d2b282dd865c6f99df6fd6a3f97ccb41727b7658#diff-29eca65bc5a52a720033aadba07e3920

I've also added a command-line argument to both `rgb-to-gif.py` and `rawrgb` to let you specify a framerate, and have lowered the default framerate to one that's more likely to work for (hopefully) most users.

And here's a huge gif of me waving at what's supposed to be 4fps:

[image]

robmarkcole, 2019-04-11:
Still a bit wild https://github.com/pimoroni/mlx90640-library/commit/d2b282dd865c6f99df6fd6a3f97ccb41727b7658#commitcomment-33139389

Gadgetoid, 2019-04-11:
Tried 1FPS? Also make sure you have nothing else on the i2c bus. Right now your data - looking at https://github.com/pimoroni/mlx90640-library/commit/d2b282dd865c6f99df6fd6a3f97ccb41727b7658#diff-29eca65bc5a52a720033aadba07e3920 - looks completely garbled.

Might be worth also running one of the C examples directly (sdlscale is one of the better ones) to see if you get the same result.

What version of the Pi is this on? How is the sensor connected to the Pi?

robmarkcole, 2019-04-11:
As below. OK I restarted the Pi and Gifs look good. However now my `rgb-to-file.py` is now displaying a blank image. Something very funny going on here 

[image]

Gadgetoid, 2019-04-11:
You'll need to solder a header on, or pick up a Breakout Garden. I2C will be an unreliable mess - as demonstrated by your camera feed - without a solid electrical connection.

Unsoldered IO pins are - literally - square pegs in round holes.

robmarkcole, 2019-04-11:
I don't think thats the problem (although I agree this needs a solder!) as I capture the gifs just fine.

Gadgetoid, 2019-04-11:
Just fine!?!?

[image]

Gadgetoid, 2019-04-11:
To be fair- I've just jammed my camera on in very much the same way and it works surprisingly well. Well kind of.

robmarkcole, 2019-04-11:
After reboot

[image]

Gadgetoid, 2019-04-11:
What the!? Wonders never cease.

robmarkcole, 2019-04-11:
OK so I saved each of the frames from the gif, the zeroth frame is blank

[image]

By frame 3 we are getting data

[image]

Frame 5 is starting to look reasonable

[image]

Gadgetoid, 2019-04-11:
I reckon a `--drop-frames` argument would be a good addition, so you can specify how much data to throw away before the recording starts. Or perhaps it should be a timed delay. Would give much better loops, as the black frames currently break that.

robmarkcole, 2019-04-11:
Nice! Thanks for your help on this one

closed 2019-04-11 by robmarkcole