pimoroni/mlx90640-library #18: can aligned the thermal value with pi camera?

koks123456, 2019-04-11:
Hi, can l save the thermal value obtained in a file and then display the value in RGB format which is aligned with the image from pi camera continuously? can you help me or give some directions for me? I'm still new in this field, hope can help me. Thank you so much

Gadgetoid, 2019-04-12:
We've done this- poorly- with a regular camera and OBS studio, but that might be overkill for a Pi.

I suggest you start here: https://github.com/pimoroni/unicorn-hat-hd/blob/master/examples/camera.py

And here: https://github.com/pimoroni/mlx90640-library/blob/master/python/rgb-to-gif.py

And look at gluing these examples together. Ideally you'd capture from the Pi camera at some multiple of the thermal sensor resolution to get 1:1 alignment of pixels.

koks123456, 2019-04-12:
thanks for your reply. For example, if l save the thermal value (val) in txt.file, but how to open and save it in array form so that l can reshape the alignment of pixels and aligned or overlap with picamera? Can give some suggestions?

Gadgetoid, 2019-04-12:
I'm considering this a feature request, since it's a good idea. But I don't think I'll get time to work on it soon.

In the interim GitHub issues aren't really the place for general programming help, since I'm more-or-less the only one looking at these and don't have the time to go into detail.

You should direct your questions over at the Pi forums: https://www.raspberrypi.org/forums/

Or our forums: https://forums.pimoroni.com/

koks123456, 2019-04-12:
Thank you. If got any ideas in the future please let me know.

closed 2019-04-12 by koks123456