pimoroni/mlx90640-library #19: Streaming over network

robmarkcole, 2019-04-11:
Not an issue, but I would love to know the 'best' way to stream the feed over a local network. Ideally I want to be able to view in a browser and optionally save frames

timsoftgit, 2019-04-11:
you might find just saving a pic and having the browser reload every 30 seconds with js is a simple way to do this. I do something similar with a regular pi cam, but not video, just frames every 30sec.

Gadgetoid, 2019-04-12:
The huge comment block on the top of `rawrgb.cpp` should help: https://github.com/pimoroni/mlx90640-library/blob/master/examples/src/rawrgb.cpp

closed 2019-04-12 by Gadgetoid

robmarkcole, 2019-04-12:
HI @Gadgetoid 
The instructions you mentioned work, although there is a funny issue with the frame being 'wrapped around', so that the right hand side of the scene is on the left (below). Any pointers on how I could fix this? Thanks!

[image]

robmarkcole, 2019-04-16:
OK this behaviour is due to the frame rate - if I drop to 8 FPS the images are normal