pimoroni/mlx90640-library #20: Example rawrgb doesn't seem to combine subframes

marcus-clements, 2019-04-15:
Probably just a lack of understanding, apologies in advance if so :)

in rawrgb.cpp line 191 we added code to print out the subpage from frame[833]
https://github.com/pimoroni/mlx90640-library/blob/master/examples/src/rawrgb.cpp#L191

Each time round the loop the subpage alternates 0,1.

According to the docs the subpages should be available at double the framerate. We thought maybe they should be combined to make a single frame, rather than using each indivdually, and this might reduce the chess-board pattern exhibited by fast moving images?

Gadgetoid, 2019-04-16:
As near as I can tell the datasheet - in the Measurement Flow section on page 13/60 (https://www.melexis.com/-/media/files/documents/datasheets/mlx90640-datasheet-melexis.pdf) - claims you must "Wait time determined by RR - 20%" between sub frames meaning that the full frame rate is, in practise, only half that of the refresh rate.

The data coming out of the sensor has zeros in these chessboard regions, so if you were seeing the raw data you'd probably notice a strobing effect between black and valid pixels.

What you're seeing is a combination of sensor noise and temporal effects that mean each sub frame, despite the fact they are combined together, is different enough from the last to betray the chessboard effect.

The `MKL90640_CalculateTo` method combines alternating sub frames together such that each new sub frame produces a new frame- but in practise only half the pixels of that new frame have been updated since the last frame and because it's not possible to have 0 time between each sub frame, they will never precisely correllate.

Possibly your best way to fix this is to run the camera at the maximum framerate and smooth out the noise over time. Running at 64FPS with a 4 sample smoothing filter might give you a fairly stable 16FPS image, but unless you set up a software I2C bus with `i2c-gpio` you're unlikely to be able to run any other devices alongside it.

marcus-clements, 2019-04-17:
Thanks, very helpful.

closed 2019-04-17 by marcus-clements