pimoroni/mlx90640-library #21: inconsistent horizontal striped line frames

koks123456, 2019-04-18:
When l run the interp.cpp, the frames are striped horizontally. l try to set different fps but nothing works. But when l test the same sensor with the evaluation board provided by Melexis (MLxCIRT 90640 3), it produces better frames without the stripes. 
Does anyone have any idea to fix it? Thank you
[image]

Gadgetoid, 2019-04-24:
I have to admit defeat on this one- I played with the sensor and C examples for several days just trying to get it to output any video at all, and I haven't figured out how to minimise the temporal/interlacing distortion effects.

I'm eager for any input on this.

erikwitkamp, 2019-06-15:
I'm not familiar with the Melexis eval board, but I've had 4 of these MLX90640 sensors connected to one RPi at a time.  One particular sensor seemed to always require a POR to not display horizontal lines, no matter what.  Then, it died.  No ACK on I2C bus.  Since, I've had another one seemingly stop working altogether.  So, they may just be finicky/delicate.  I may also just be inept!

Prior to that, two of these same four would not display at all, while the other two would work fine with the same code.  Dead pixel compensation is supposed to be taken care of, but the sensor was putting out NULL for these pixels, instead of zero (or some other value).  When the NULL value was the very last value of a frame, the code broke.  Replacing the NULL values with the row average was a quick way to fix it.