pimoroni/mlx90640-library #22: Segmentation Fault

gordon227, 2019-05-28:
Afternoon, 
So i have tried everything over 4 times i have ran i2cdetect -y 1 i get it at 33 but everytime i sudo examples/fbuf or any other i get a segmentation fault i tried to debug but i can not figure out why its failing. I restarted from the top of the readme and went through it step by step. Anyone else experience this?

pi@ThemalPi:~/Documents/mlx90640-library-master $ sudo examples/interp
The framebuffer device was opened successfully.
Original 1920x1080, 32bpp
Segmentation fault

When i run rawrgb it will just sit there no output or anything which makes me wonder if it is trying to open a window or something but usually you get a response on the CLI. 

If i make examples/video i get this out put 
`g++ -I. -std=c++11 -std=c++11   -c -o examples/src/video.o examples/src/video.cpp
examples/src/video.cpp: In function ‘void video_encode_start(const char*, int, AVCodecID)’:
examples/src/video.cpp:118:25: warning: ‘AVStream::codec’ is deprecated [-Wdeprecated-declarations]
     if (video_avstream->codec == NULL) {
                         ^~~~~
In file included from examples/src/video.cpp:21:0:
/usr/include/arm-linux-gnueabihf/libavformat/avformat.h:893:21: note: declared here
     AVCodecContext *codec;
                     ^~~~~
examples/src/video.cpp:118:25: warning: ‘AVStream::codec’ is deprecated [-Wdeprecated-declarations]
     if (video_avstream->codec == NULL) {
                         ^~~~~
In file included from examples/src/video.cpp:21:0:
/usr/include/arm-linux-gnueabihf/libavformat/avformat.h:893:21: note: declared here
     AVCodecContext *codec;
                     ^~~~~
examples/src/video.cpp:118:25: warning: ‘AVStream::codec’ is deprecated [-Wdeprecated-declarations]
     if (video_avstream->codec == NULL) {
                         ^~~~~
In file included from examples/src/video.cpp:21:0:
/usr/include/arm-linux-gnueabihf/libavformat/avformat.h:893:21: note: declared here
     AVCodecContext *codec;
                     ^~~~~
examples/src/video.cpp:124:25: warning: ‘AVStream::codec’ is deprecated [-Wdeprecated-declarations]
     c = video_avstream->codec;
                         ^~~~~
In file included from examples/src/video.cpp:21:0:
/usr/include/arm-linux-gnueabihf/libavformat/avformat.h:893:21: note: declared here
     AVCodecContext *codec;
                     ^~~~~
examples/src/video.cpp:124:25: warning: ‘AVStream::codec’ is deprecated [-Wdeprecated-declarations]
     c = video_avstream->codec;
                         ^~~~~
In file included from examples/src/video.cpp:21:0:
/usr/include/arm-linux-gnueabihf/libavformat/avformat.h:893:21: note: declared here
     AVCodecContext *codec;
                     ^~~~~
examples/src/video.cpp:124:25: warning: ‘AVStream::codec’ is deprecated [-Wdeprecated-declarations]
     c = video_avstream->codec;
                         ^~~~~
In file included from examples/src/video.cpp:21:0:
/usr/include/arm-linux-gnueabihf/libavformat/avformat.h:893:21: note: declared here
     AVCodecContext *codec;
                     ^~~~~
examples/src/video.cpp: In function ‘void video_encode_frame(int)’:
examples/src/video.cpp:207:11: warning: ‘int avcodec_encode_video2(AVCodecContext*, AVPacket*, const AVFrame*, int*)’ is deprecated [-Wdeprecated-declarations]
     ret = avcodec_encode_video2(c, &pkt, frame, &got_output);
           ^~~~~~~~~~~~~~~~~~~~~
In file included from examples/src/video.cpp:15:0:
/usr/include/arm-linux-gnueabihf/libavcodec/avcodec.h:5333:5: note: declared here
 int avcodec_encode_video2(AVCodecContext *avctx, AVPacket *avpkt,
     ^~~~~~~~~~~~~~~~~~~~~
examples/src/video.cpp:207:60: warning: ‘int avcodec_encode_video2(AVCodecContext*, AVPacket*, const AVFrame*, int*)’ is deprecated [-Wdeprecated-declarations]
     ret = avcodec_encode_video2(c, &pkt, frame, &got_output);
                                                            ^
In file included from examples/src/video.cpp:15:0:
/usr/include/arm-linux-gnueabihf/libavcodec/avcodec.h:5333:5: note: declared here
 int avcodec_encode_video2(AVCodecContext *avctx, AVPacket *avpkt,
     ^~~~~~~~~~~~~~~~~~~~~
examples/src/video.cpp:221:3: warning: ‘void av_free_packet(AVPacket*)’ is deprecated [-Wdeprecated-declarations]
   av_free_packet(&pkt);
   ^~~~~~~~~~~~~~
In file included from examples/src/video.cpp:15:0:
/usr/include/arm-linux-gnueabihf/libavcodec/avcodec.h:4483:6: note: declared here
 void av_free_packet(AVPacket *pkt);
      ^~~~~~~~~~~~~~
examples/src/video.cpp:221:22: warning: ‘void av_free_packet(AVPacket*)’ is deprecated [-Wdeprecated-declarations]
   av_free_packet(&pkt);
                      ^
In file included from examples/src/video.cpp:15:0:
/usr/include/arm-linux-gnueabihf/libavcodec/avcodec.h:4483:6: note: declared here
 void av_free_packet(AVPacket *pkt);
      ^~~~~~~~~~~~~~
examples/src/video.cpp: In function ‘void video_encode_end()’:
examples/src/video.cpp:242:48: warning: ‘AVStream::codec’ is deprecated [-Wdeprecated-declarations]
         av_freep(&avFormatContext->streams[i]->codec);
                                                ^~~~~
In file included from examples/src/video.cpp:21:0:
/usr/include/arm-linux-gnueabihf/libavformat/avformat.h:893:21: note: declared here
     AVCodecContext *codec;
                     ^~~~~
examples/src/video.cpp:242:48: warning: ‘AVStream::codec’ is deprecated [-Wdeprecated-declarations]
         av_freep(&avFormatContext->streams[i]->codec);
                                                ^~~~~
In file included from examples/src/video.cpp:21:0:
/usr/include/arm-linux-gnueabihf/libavformat/avformat.h:893:21: note: declared here
     AVCodecContext *codec;
                     ^~~~~
examples/src/video.cpp:242:48: warning: ‘AVStream::codec’ is deprecated [-Wdeprecated-declarations]
         av_freep(&avFormatContext->streams[i]->codec);
                                                ^~~~~
In file included from examples/src/video.cpp:21:0:
/usr/include/arm-linux-gnueabihf/libavformat/avformat.h:893:21: note: declared here
     AVCodecContext *codec;
                     ^~~~~
examples/src/video.cpp: In function ‘void video_encode_start(const char*, int, AVCodecID)’:
examples/src/video.cpp:195:49: warning: ignoring return value of ‘int avformat_write_header(AVFormatContext*, AVDictionary**)’, declared with attribute warn_unused_result [-Wunused-result]
  avformat_write_header(avFormatContext, &params);
                                                 ^
g++ -L/home/pi/mlx90640-library examples/src/video.o examples/src/lib/fb.o libMLX90640_API.a -o examples/video -lbcm2835 -lavcodec -lavutil -lavformat -lbcm2835
`

Then when i run sudo examples/video the output is

`pi@ThemalPi:~/Documents/mlx90640-library-master $ sudo examples/video 
The framebuffer device was opened successfully.
Original 1920x1080, 32bpp
Setting output format...
Setting filename...
Add video streamt...
Opening codec...
Allocating image...
Writing header...
[apng @ 0x2e0770] Using AVStream.codec.time_base as a timebase hint to the muxer is deprecated. Set AVStream.time_base instead.
[apng @ 0x2e0770] Using AVStream.codec to pass codec parameters to muxers is deprecated, use AVStream.codecpar instead.
Encode started..
got_output false: 0Write frame   1 (size=   75)
Write frame   2 (size=   66)
Write frame   3 (size=   66)
Write frame   4 (size=   66)
Write frame   5 (size=   66)
Write frame   6 (size=   66)
Write frame   7 (size=   66)
Write frame   8 (size=   66)
........

Write frame  76 (size=   66)
Write frame  77 (size=   66)
Write frame  78 (size=   66)
Write frame  79 (size=   66)
*** Error in `examples/video': double free or corruption (!prev): 0x002e10f0 ***
Aborted
pi@ThemalPi:~/Documents/mlx90640-library-master $ ^C

`
I am baffled if anyone can assist me in figuring this out i would greatly appreciate it.

Thanks

Gadgetoid, 2019-06-09:
I haven't had a chance to look into either yet, but I wonder if this is related to https://github.com/pimoroni/mlx90640-library/issues/24

gordon227, 2019-06-09:
Well I was using the spark fun break out and I bit the bullet and bought the pimoroni every example works except video and one more can't remember which one I'll update in a little while but still get the same error I have not compared the 2 different breakouts to see what the differences are. Which might be the actual cause for all of the examples not working.

Gadgetoid, 2019-06-11:
I'm guessing it's possible that BadPixelsCorrection might segfault when there are no bad pixels to correct. The underlying API code is not especially robust.

It looks like `brokenPixels` and `outlierPixels` are initialised to `0xFFFF` but I'll have to dig a little deeper to find out why this breaks.

In fact I can replicate your SegFault by removing the code that loads bad/outlier pixels from the device.

Gadgetoid, 2019-06-11:
Okay! See https://github.com/pimoroni/mlx90640-library/commit/93bb625bcd1a6269738459a85483d6057f2d6d7a#diff-26f614063f9bb2756155e3e26d51b772R120 and see if it fixes your segfault issue.

I'm guessing the difference is that on the SparkFun board you won the pixel lottery and got a breakout with no deviating or dead pixels, whereas ours had at least one.

This might be an issue worth reporting upstream since it means the `MLX90640_BadPixelsCorrection` function doesn't actually check to see if there's anything to correct. There's already enough boilerplate checks and balances in the consuming code (fbuf.cpp) without having yet more of it so I think it's reasonable to expect that the API handles this.

gordon227, 2019-06-12:
I will give this a try tomorrow morning for sure and report back

roblevine, 2019-10-12:
Did you ever resolve this? I am having exactly the same problem on my PI3.

Gadgetoid, 2019-11-07:
This fix should have resolved the BadPixelsCorrection segfault- https://github.com/pimoroni/mlx90640-library/commit/f4549dc6cc416f8db05a8bfd1d4850aee5e7af75#diff-ddebbbc992600504b2034532b938838b

Which was discussed upstream here: https://github.com/melexis/mlx90640-library/issues/38

closed 2020-03-16 by Gadgetoid