pimoroni/mlx90640-library #23: Checkerboard pattern randomly

nunomiguelferreira, 2019-05-30:
Hello,

So I'm using this library to get a reading every minute, but sometimes after some hours of readings I get this checkerboard pattern that you can observe in this apng.

[video_checker](https://user-images.githubusercontent.com/33719583/58631224-c835e480-82d9-11e9-8851-05a5557df81e.gif)

The only way to make the library work again is to reboot the pi. So I'm wondering, if you know if the problem is from the camera being out of sync (which would be weird, since a new ./video is run every minute) or if the problem stems from somewhere in the I2C bus.

I'd also like to say that I'm writing the array temperature values for each frame read in a text file and this is with a RPI3
!

Thank you for your time

Gadgetoid, 2019-06-04:
I've not seen this myself- but I also haven't run the camera for more than a few hours. It's a very finickity sensor that I've had precious little luck bending to my will- not for want of trying.

Unfortunately I've got too much on my plate to give this a proper look at the moment, but it's been on my TODO list for a while.

nunomiguelferreira, 2019-06-09:
Yeah exactly, I only run into this problem when running the sensor for over 16hours at a time. I've managed to fix it by cutting power to the RPI board once every 8 hours, but it is not the most elegant solution for a fix.

Thank you for your input!

dglaude, 2019-12-28:
Checkboard pattern is the way this sensor work. It take two half frame with alternate checkboard.
You would notice that with moving object.
This library might hide this, but then over time something goes wrong.
You could try with Adafruit new CircuitPython library with Blinks.

Technofinchen, 2020-02-20:
Having the same problem trying to pipe the rawrgb.cpp output to flask without buffering. The problems exists permanently. Does the rawrgb script generate a full 24x32 rgb matrix?

...
`try:
		with subprocess.Popen(["sudo", trgbpath, "{}".format(fps)], stdin=subprocess.PIPE, stdout=subprocess.PIPE, stderr=subprocess.PIPE) as tcamera:
			while True:
					tframe = tcamera.stdout.read(2304)
					size = len(tframe)
					print("Got {} bytes of data!".format(size))
					image = Image.frombytes('RGB', (32, 24), tframe)

					break`
...

closed 2023-01-20 by Gadgetoid