pimoroni/mlx90640-library #26: Issue with running MLX90640 and BME680 simultaneously

sylchw, 2019-07-05:
Hi, I tried to run the example codes of the BME680 (read-all.py) and MLX90640 (hotspot.cpp) together in parallel on the raspberry pi, but both crashes upon attempting to run them simultaneously.

The error shown on the read-all.py script is this:
Traceback (most recent call last):
File "read-all.py", line 58, in 
if sensor.get_sensor_data():
File "build/bdist.linux-armv7l/egg/bme680/init.py", line 264, in get_sensor_data
File "build/bdist.linux-armv7l/egg/bme680/init.py", line 248, in set_power_mode
File "build/bdist.linux-armv7l/egg/bme680/init.py", line 304, in _set_bits
File "build/bdist.linux-armv7l/egg/bme680/init.py", line 319, in _get_regs
IOError: [Errno 5] Input/output error

From what I'm aware, they're on different i2c addresses, with the MLX90640 at 0x33 and BME680 at 0x76.

Does anyone know how this issue can be resolved?

Thank you

Gadgetoid, 2019-07-05:
What speed are you running the i2c bus for the MLX90640?

Bear in mind that I2C is also not thread-safe, so if you run the two scripts together with nothing to prevent them both trying to talk over I2C then you're going to end up with a mess of interleaved, half-complete commands at best.

Gadgetoid, 2019-11-08:
It might be worth using `i2c-gpio` to set up another software i2c bus and connect the BME680 - and other less troublesome devices - to that bus instead of the main one.

Find usage information here: 

https://github.com/raspberrypi/linux/blob/f555d78ee33b13b8eaf21d6e191a7bc68cee8a14/arch/arm/boot/dts/overlays/README#L1017-L1032

```
Name:   i2c-gpio
Info:   Adds support for software i2c controller on gpio pins
Load:   dtoverlay=i2c-gpio,<param>=<val>
Params: i2c_gpio_sda            GPIO used for I2C data (default "23")

        i2c_gpio_scl            GPIO used for I2C clock (default "24")

        i2c_gpio_delay_us       Clock delay in microseconds
                                (default "2" = ~100kHz)

        bus                     Set to a unique, non-zero value if wanting
                                multiple i2c-gpio busses. If set, will be used
                                as the preferred bus number (/dev/i2c-<n>). If
                                not set, the default value is 0, but the bus
                                number will be dynamically assigned - probably
                                3.
```

And for more details see: https://github.com/raspberrypi/linux/blob/rpi-4.19.y/Documentation/devicetree/bindings/i2c/i2c-gpio.txt
And: https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/i2c-gpio-overlay.dts