pimoroni/mlx90640-library #32: Camera I2C issues

xbst, 2019-09-21:
I am using MLX90640 IR camera with my PI 3 B. The camera is on PIM366 from digikey. The problem I am is with I2C. I checked the i2c configurations which is 0x33 and checked with i2cdetect on pi console and got the address. Later I started working on getting libraries and codes setup. Now I am having errors with I2C. Its not detecting any address. When I compile everything I get this error "frameData timeout error waiting for dataready. Failed to get frame data. I have the SDA and SCA connected to pin 3 and 5 of pi. I doubt its the coonections issues but couldn't find what it is.

Gadgetoid, 2019-09-23:
The MLX90640 is also extremely timing sensitive-

1. Are you using the Python or C libraries
2. What framerate are you trying to run the camera at
3. How is it wired to SDA/SCL?
4. Is there anything else on that i2c bus.

xbst, 2019-09-24:
Are you using the Python or C libraries
-> I am using the C library.
What framerate are you trying to run the camera at
-> I am working on the 8 FPS, as unchanged from this library I cloned,
How is it wired to SDA/SCL?
-> The SDA/SCL are conneced directly to pin 3 and 5 of Pi.
Is there anything else on that i2c bus.
-> No there is nothing on i2c bus.  I am getting I2C read error this time.

closed 2019-10-25 by SushanTiwari