pimoroni/mlx90640-library #36: Error in test.py (in python experimental library): undefined symbol: cleanup

smartyram, 2019-11-18:
I get this error when running test.py in in python experimental library on Raspberry Pi 4

```Traceback (most recent call last):
  File "/home/pi/Documents/python/thermal/thermal_lib/MLX90640.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 670, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 583, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1043, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: /home/pi/Documents/python/thermal/venv/lib/python3.7/site-packages/MLX90640-0.0.2-py3.7-linux-armv7l.egg/_MLX90640.cpython-37m-arm-linux-gnueabihf.so: undefined symbol: cleanup

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test.py", line 6, in <module>
    import MLX90640 as x
  File "/home/pi/Documents/python/thermal/thermal_lib/MLX90640.py", line 17, in <module>
    _MLX90640 = swig_import_helper()
  File "/home/pi/Documents/python/thermal/thermal_lib/MLX90640.py", line 16, in swig_import_helper
    return importlib.import_module('_MLX90640')
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: /home/pi/Documents/python/thermal/venv/lib/python3.7/site-packages/MLX90640-0.0.2-py3.7-linux-armv7l.egg/_MLX90640.cpython-37m-arm-linux-gnueabihf.so: undefined symbol: cleanup```

owenrog, 2019-11-23:
I am getting exactly the same error on RPi3.

Chetchita641, 2019-11-26:
The issue is that the linker is using C naming conventions for a C++ file. To fix this go in mlx90640-python.cpp and you need to uncomment the 'extern "C"' from each function and wrap each one in brackets, or the whole thing, like so:

`extern "C"
{
 ...
}`

smartyram, 2019-11-26:
@Chetchita641  thank you for replying
> The issue is that the linker is using C naming conventions for a C++ file

Is there a linker switch to prevent it from doing this? Can we change the make file to prevent this from happening?
I am wondering how the original author built this library?

Gadgetoid, 2020-03-09:
Sorry for the delay picking this up - the Python library has always been something of a secondary concern since I never could make it produce satisfactory results.

Hopefully this is fixed in https://github.com/pimoroni/mlx90640-library/commit/6c573af320cee650071d52d65402bd86aa17fd51 which uses SWIG opts to force the output file to be C++ rather than C. Local tests seem to suggest this is the case, although I'm more often than not tripping over preinstalled modules from previous tests and ending up with false positves.

Gadgetoid, 2020-03-10:
Did this fix work in your case?

closed 2020-03-16 by Gadgetoid