pimoroni/mlx90640-library #37: `AttributeError: __exit__` when creating subprocess in sample python code

mylogon341, 2019-11-30:
Hi. I'm very new to python (currently a lead ios dev) and have little experience in debugging (beyond google).  I am trying to run the `rgb-to-gif.py` script on a pi zero w (headless). Building, earlier on, i used the `make I2C_MODE=RPI` method. I believe it is using python2.7 by default.
When I try to run it, I just end up hitting this error, which i'm having trouble finding help with elsewhere on the ole' internet.  
I've literally changed nothing in the code, and it looks like I successfully built the `rawrgb` correctly. But if no one else is having this issue, I would suggest i've skimmed over an important detail somewhere 🤔 

``` bash
Traceback (most recent call last):
  File "rgb-to-gif.py", line 43, in <module>
    with subprocess.Popen(["sudo", RAW_RGB_PATH, "{}".format(fps)], stdin=subprocess.PIPE, stdout=subprocess.PIPE, stderr=subprocess.PIPE) as camera:
AttributeError: __exit__
``` 

Thanks

strangerman, 2019-12-25:
I'm having the same issue. Were you able to find a solution for this?

Gadgetoid, 2020-03-10:
This is an obtuse error and no mistake. Python classes are supposed to define `__enter__` and `__exit__` methods which allow for `with X as Y:` to work. `__enter__` is called at the start of the `with`, and `__exit__` is called when it goes out of scope.

It's possible that in Python 2.x these attributes aren't defined on `subprocess.Popen`. In fact this very bug is described here: https://bugs.python.org/issue13202

Looks like context management support (the `__enter__` and `__exit__` methods) was not added to `subprocess.Popen` until Python 3.2.