pimoroni/mlx90640-library #39: Raspberry Pi4 I2C Issues

S4WRXTTCS, 2019-12-18:
When I run any of the examples I get " frameData timeout error waiting for dataready" errors. Depending on the examples this either ends in a segmentation fault or some other error/crash.

I've tried lowering the framerate, but that doesn't seem to solve the issue.

One thing that is odd is the I2C Clock speed isn't actually 400KHz, but around 165KHz when viewed on the scope. From google searching I came across some information that said you had to set the core_freq to 250 in order to lock the I2C speed down, but trying that resulted in an I2C speed of 333KHz which also didn't work.

Regardless of the I2C speed the I2Cdetect utility is able to find the sensor just fine at address 0x33 so the sensor is detected, and is wired correctly.

Gadgetoid, 2019-12-18:
What version of Raspberry Pi and accompanying operating system are you using?

It might be worth shooting for a higher speed by adding this into `/boot/config.txt` (albeit this will often interfere with other devices that don't support this speed):

```
dtparam=i2c_arm_baudrate=1000000
```

S4WRXTTCS, 2019-12-18:
I'm using the Raspberry PI 4B with 4GB of memory. I'm using Raspian, but I'm not sure the version. If I do uname -r it reports back 4.19.75-v7l. 

I did try using the 1M baud rate, and it still didn't work. 

Then I tried attaching the Pimoroni MLX90640 board to the Melexis MLX90640-41 EVB Board to see if there wasn't an issue with the Sensor itself. It did detect it just fine, but it gave nothing for the thermal image. On that board it seems to be hard coded to 1M for the I2C Speed.  Now it's possible that pull ups on the Pimoroni board were messing things up or something else. For whatever reason that sensor doesn't seem to work correctly. 

So then I tried another MLX90640 sensor from the MLX90640-41 EVB Board on the Raspberry PI, and that doesn't have the frameData timeout errors. 

But, the video example fails with "double free or corruption (!prev)", and then "aborted" 

Here is the output with the frame rate set at 1fps. 

pi@raspberrypi:~/Rasp-I2C/thermal/mlx90640-library $ ./examples/video 
The framebuffer device was opened successfully.
Original 1920x1080, 32bpp
Setting output format...
Setting filename...
Add video streamt...
Opening codec...
Allocating image...
Writing header...
[apng @ 0x16e0b00] Using AVStream.codec to pass codec parameters to muxers is deprecated, use AVStream.codecpar instead.
Encode started..
got_output false: 0Write frame   1 (size=  792)
Write frame   2 (size= 1262)
[apng @ 0x16e0b00] Frame rate is too high or specified too precisely. Unable to copy losslessly.
Write frame   3 (size= 1254)
Write frame   4 (size= 1263)
double free or corruption (!prev)
Aborted

sneakers-the-rat, 2020-01-08:
I don't write C++ so I'm not sure if I'm right, but I think that if you make it like

`make all I2C_MODE=RPI`, then it hardcodes the baudrate at 400000 (see functions/MLX90640_RPI_I2C_Driver.cpp). The baudrate set in /python/library/mlx90640-python.cpp never seems to be used.

I forked and changed [this line](https://github.com/pimoroni/mlx90640-library/blob/bac70fab79896fdc8bcf41c3f55b8c01fab72e13/functions/MLX90640_RPI_I2C_Driver.cpp#L33) so that the baudrate was 1000000 (after also changing /boot/config.txt) and have been able to capture at 64fps just fine.

closed 2023-01-20 by Gadgetoid