pimoroni/mlx90640-library #42: ModuleNotFoundError: No module named 'MLX90640

Charlenesy, 2020-02-19:
For this code : https://github.com/pimoroni/mlx90640-library

Unable to find the module MLX90640 in the code given.
I have already installed the library, but the following error was given:
'ModuleNotFoundError: No module named 'MLX90640'

Kindly advise how to resolve this issue?

`import sys
import colorsys
from PIL import Image
sys.path.insert(0, "./build/lib.linux-armv7l-2.7")

import MLX90640 as x

img = Image.new( 'RGB', (24,32), "black")

def temp_to_col(val):
    hue = (180 - (val * 6)) / 360.0
    return tuple([int(c*255) for c in colorsys.hsv_to_rgb(hue % 1, 1.0, 1.0)])

# def temp_to_col(val):
#     return tuple([int(c*255) for c in colorsys.hsv_to_rgb(1.0, 0.0, val/50.0)])
`

Gadgetoid, 2020-02-23:
The Python library is a little rough around the edges and probably not much use, but make sure you go into `https://github.com/pimoroni/mlx90640-library/tree/master/python/library` and `make build && make install`.

Gadgetoid, 2020-03-09:
Any luck with this? Feel free to re-open if you're still having issues.

closed 2020-03-16 by Gadgetoid