pimoroni/mlx90640-library #44: Save as png or jpg

Jooslee0313, 2020-02-26:
Hi, 
how can I save as png or jpg by using rbg-to-gif.py ? 

and When I use the compiled step file like this ./step >> /home/pi/Desktop/123.txt

some result values were displayed wrong numbers.
ex>
 Starting...
[Datapoint=1] 
Subpage: 0
+00.00 +154.67 +00.00 +194.23 +00.00 +218.49 +00.00 +238.84 +00.00 +244.62 +00.00 +254.52 +00.00 +259.41 +00.00 +258.42 +00.00 +256.24 +00.00 +251.47 +00.00 +234.59 +00.00 +173.17 
this is first line of 32x24 data matrix.

I hope answer my questions. 
Thank you

Gadgetoid, 2020-02-26:
`./step` is pretty useless, since it iterates through the sensor data pages one-by-one and does not (if I remember correctly) attempt to do interpolation. Your dumped file is raw data from the sensor, and is coming out exactly how I would expect it- each odd/even frame has a different set of pixels activated in either a scanline or chessboard pattern, and these must be combined to get a full frame of data.

From experience the sensor startup time means you might need to skip a few frames before saving one out as png or jpg.

`rgb-to-gif.py` has each frame as an individual image from:

https://github.com/pimoroni/mlx90640-library/blob/6fd7791d6b180d998b79e9b86b60ac79b9a525f1/python/rgb-to-gif.py#L56-L57

This image object can be saved on its own, and you can drop all of the code that collects frames and converts the result to a gif. See the Pillow docs: https://pillow.readthedocs.io/en/3.1.x/reference/Image.html#PIL.Image.Image.save

Jooslee0313, 2020-02-26:
Thank you for your reply.

First, my explanation made confusion. So, I attached step.cpp file which I
modify for me and extracted data file.
Open the data file and you can find some strange data. for example,
datapoint 14 is strange.
I got 240 data by using ./step >> /home/pi/123.txt. there are 50 strange
data.
How can I fix this problem?

Second, how can I change L56-L57 ? Actually, I'm very sorry to ask you.  I
really need your help.

I hope positive answer from you soon

Thank you,
Joosang

Starting...
[Datapoint=1] 
Subpage: 0
+00.00 +154.67 +00.00 +194.23 +00.00 +218.49 +00.00 +238.84 +00.00 +244.62 +00.00 +254.52 +00.00 +259.41 +00.00 +258.42 +00.00 +256.24 +00.00 +251.47 +00.00 +234.59 +00.00 +173.17 
+146.53 +00.00 +211.57 +00.00 +233.26 +00.00 +252.74 +00.00 +262.33 +00.00 +267.46 +00.00 +271.53 +00.00 +271.23 +00.00 +272.18 +00.00 +268.53 +00.00 +266.88 +00.00 +244.66 +00.00 
+00.00 +199.71 +00.00 +223.97 +00.00 +250.87 +00.00 +263.05 +00.00 +269.97 +00.00 +274.21 +00.00 +281.13 +00.00 +277.27 +00.00 +272.59 +00.00 +266.98 +00.00 +259.81 +00.00 +255.53 
+191.66 +00.00 +235.20 +00.00 +259.24 +00.00 +267.63 +00.00 +272.34 +00.00 +279.97 +00.00 +287.99 +00.00 +286.31 +00.00 +286.39 +00.00 +285.32 +00.00 +277.91 +00.00 +267.82 +00.00 
+00.00 +235.22 +00.00 +244.91 +00.00 +267.16 +00.00 +275.80 +00.00 +279.01 +00.00 +287.58 +00.00 +295.63 +00.00 +286.12 +00.00 +290.27 +00.00 +286.02 +00.00 +276.18 +00.00 +267.73 
+225.29 +00.00 +258.22 +00.00 +271.95 +00.00 +282.00 +00.00 +287.83 +00.00 +293.73 +00.00 +295.30 +00.00 +294.64 +00.00 +297.13 +00.00 +297.60 +00.00 +287.79 +00.00 +283.71 +00.00 
+00.00 +248.48 +00.00 +260.84 +00.00 +277.13 +00.00 +285.21 +00.00 +289.57 +00.00 +293.92 +00.00 +296.74 +00.00 +309.23 +00.00 +298.21 +00.00 +291.92 +00.00 +289.71 +00.00 +279.77 
+240.97 +00.00 +265.35 +00.00 +282.36 +00.00 +292.08 +00.00 +296.21 +00.00 +309.45 +00.00 +304.27 +00.00 +305.78 +00.00 +302.14 +00.00 +300.23 +00.00 +296.97 +00.00 +291.66 +00.00 
+00.00 +266.18 +00.00 +276.44 +00.00 +288.44 +00.00 +297.94 +00.00 +300.86 +00.00 +306.54 +00.00 +303.94 +00.00 +301.99 +00.00 +302.09 +00.00 +304.38 +00.00 +297.65 +00.00 +289.18 
+251.00 +00.00 +277.72 +00.00 +290.75 +00.00 +297.52 +00.00 +302.19 +00.00 +308.49 +00.00 +311.77 +00.00 +309.72 +00.00 +305.55 +00.00 +303.76 +00.00 +303.20 +00.00 +298.13 +00.00 
+00.00 +267.17 +00.00 +283.23 +00.00 +294.08 +00.00 +305.10 +00.00 +309.21 +00.00 +315.06 +00.00 +307.90 +00.00 +308.38 +00.00 +306.68 +00.00 +306.43 +00.00 +297.95 +00.00 +294.08 
+257.72 +00.00 +286.55 +00.00 +296.99 +00.00 +306.67 +00.00 +312.45 +00.00 +315.84 +00.00 +316.42 +00.00 +316.13 +00.00 +311.91 +00.00 +309.59 +00.00 +308.72 +00.00 +301.95 +00.00 
+00.00 +276.85 +00.00 +286.77 +00.00 +299.86 +00.00 +313.61 +00.00 +314.90 +00.00 +320.63 +00.00 +316.93 +00.00 +313.48 +00.00 +309.54 +00.00 +308.32 +00.00 +305.81 +00.00 +296.59 
+264.84 +00.00 +287.37 +00.00 +299.33 +00.00 +307.22 +00.00 +320.30 +00.00 +321.41 +00.00 +319.79 +00.00 +313.10 +00.00 +315.39 +00.00 +309.76 +00.00 +309.75 +00.00 +303.53 +00.00 
+00.00 +276.51 +00.00 +290.09 +00.00 +300.45 +00.00 +305.27 +00.00 +315.39 +00.00 +317.63 +00.00 +310.67 +00.00 +315.34 +00.00 +313.62 +00.00 +310.44 +00.00 +305.93 +00.00 +300.51 
+266.80 +00.00 +291.30 +00.00 +301.40 +00.00 +305.69 +00.00 +313.38 +00.00 +315.38 +00.00 +314.33 +00.00 +315.11 +00.00 +314.08 +00.00 +311.50 +00.00 +309.18 +00.00 +305.37 +00.00 
+00.00 +286.44 +00.00 +294.31 +00.00 +299.23 +00.00 +311.31 +00.00 +311.16 +00.00 +314.28 +00.00 +315.95 +00.00 +315.07 +00.00 +314.23 +00.00 +314.19 +00.00 +306.58 +00.00 +302.68 
+268.57 +00.00 +289.40 +00.00 +299.46 +00.00 +305.69 +00.00 +311.45 +00.00 +311.79 +00.00 +315.61 +00.00 +316.37 +00.00 +313.28 +00.00 +313.69 +00.00 +310.22 +00.00 +303.54 +00.00 
+00.00 +280.11 +00.00 +291.56 +00.00 +299.47 +00.00 +305.90 +00.00 +308.51 +00.00 +311.38 +00.00 +313.18 +00.00 +311.96 +00.00 +314.65 +00.00 +311.84 +00.00 +306.84 +00.00 +300.17 
+268.85 +00.00 +290.56 +00.00 +299.26 +00.00 +307.23 +00.00 +308.91 +00.00 +311.51 +00.00 +314.11 +00.00 +314.24 +00.00 +312.30 +00.00 +310.94 +00.00 +306.06 +00.00 +298.29 +00.00 
+00.00 +282.25 +00.00 +293.18 +00.00 +302.05 +00.00 +307.54 +00.00 +309.99 +00.00 +311.05 +00.00 +313.59 +00.00 +312.52 +00.00 +312.16 +00.00 +308.84 +00.00 +302.36 +00.00 +298.78 
+262.46 +00.00 +284.55 +00.00 +295.61 +00.00 +302.50 +00.00 +306.83 +00.00 +310.20 +00.00 +310.41 +00.00 +310.81 +00.00 +309.45 +00.00 +308.53 +00.00 +302.79 +00.00 +300.27 +00.00 
+00.00 +274.77 +00.00 +281.46 +00.00 +291.33 +00.00 +297.01 +00.00 +304.81 +00.00 +307.79 +00.00 +308.95 +00.00 +308.68 +00.00 +307.53 +00.00 +304.53 +00.00 +300.59 +00.00 +289.58 
+251.76 +00.00 +276.50 +00.00 +291.20 +00.00 +295.85 +00.00 +303.37 +00.00 +305.38 +00.00 +307.99 +00.00 +308.46 +00.00 +307.37 +00.00 +302.49 +00.00 +298.79 +00.00 +290.33 +00.00 
+00.00 +270.43 +00.00 +280.74 +00.00 +290.54 +00.00 +296.63 +00.00 +302.61 +00.00 +306.35 +00.00 +307.78 +00.00 +307.30 +00.00 +306.24 +00.00 +300.42 +00.00 +294.33 +00.00 +289.39 
+244.37 +00.00 +266.93 +00.00 +283.90 +00.00 +293.28 +00.00 +296.68 +00.00 +299.01 +00.00 +301.26 +00.00 +301.28 +00.00 +300.66 +00.00 +296.64 +00.00 +289.94 +00.00 +280.67 +00.00 
+00.00 +249.70 +00.00 +264.18 +00.00 +280.12 +00.00 +287.61 +00.00 +291.42 +00.00 +297.34 +00.00 +296.47 +00.00 +298.40 +00.00 +291.39 +00.00 +292.69 +00.00 +284.53 +00.00 +274.38 
[... cut ...]
+278.50 +279.27 +292.30 +292.06 +297.37 +296.12 +301.40 +299.26 +304.05 +301.39 +306.22 +304.02 +308.01 +303.07 +309.92 +304.82 +307.88 +304.25 +308.71 +300.12 +304.77 +299.31 +299.40 +290.24 
+263.62 +279.38 +279.62 +286.55 +286.54 +292.56 +293.85 +296.33 +297.30 +298.99 +296.58 +299.35 +298.14 +297.08 +297.34 +298.90 +295.28 +296.55 +296.08 +294.22 +289.47 +289.73 +286.15 +276.63 
[33A[Datapoint=12] 
Subpage: 1
+21.54 +261.70 +21.62 +273.05 +20.85 +282.00 +21.67 +289.89 +20.42 +291.60 +20.36 +295.68 +21.26 +298.92 +20.86 +297.23 +21.58 +295.76 +21.74 +295.06 +21.87 +288.83 +21.01 +267

closed 2023-01-20 by Gadgetoid