pimoroni/mlx90640-library #46: RP4 cnt > 4 error

xk42, 2020-03-07:
I think I might have bricked it. I used `make I2C_MODE=LINUX` as RPI was throwing errors. Any ideas?

```
root@pi4touch:~/mlx90640-library# make I2C_MODE=RPI
g++ -fPIC -I headers -shared    -c -o functions/MLX90640_API.o functions/MLX90640_API.cpp
functions/MLX90640_API.cpp: In function ‘int MLX90640_GetData(uint8_t, uint16_t*)’:
functions/MLX90640_API.cpp:87:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
g++ -fPIC -I headers -shared    -c -o functions/MLX90640_RPI_I2C_Driver.o functions/MLX90640_RPI_I2C_Driver.cpp
ar rcs libMLX90640_API.a functions/MLX90640_API.o functions/MLX90640_RPI_I2C_Driver.o
ranlib libMLX90640_API.a
g++ -fPIC -shared functions/MLX90640_API.o functions/MLX90640_RPI_I2C_Driver.o -o libMLX90640_API.so 
g++ -I. -std=c++11 -std=c++11   -c -o examples/src/test.o examples/src/test.cpp
g++ -L/home/pi/mlx90640-library examples/src/test.o libMLX90640_API.a -o examples/test 
/usr/bin/ld: libMLX90640_API.a(MLX90640_RPI_I2C_Driver.o): in function `MLX90640_I2CRead(unsigned char, unsigned short, unsigned short, unsigned short*)':
MLX90640_RPI_I2C_Driver.cpp:(.text+0x84): undefined reference to `bcm2835_init'
/usr/bin/ld: MLX90640_RPI_I2C_Driver.cpp:(.text+0x88): undefined reference to `bcm2835_i2c_begin'
/usr/bin/ld: MLX90640_RPI_I2C_Driver.cpp:(.text+0x90): undefined reference to `bcm2835_i2c_set_baudrate'
/usr/bin/ld: MLX90640_RPI_I2C_Driver.cpp:(.text+0xe0): undefined reference to `bcm2835_i2c_setSlaveAddress'
/usr/bin/ld: MLX90640_RPI_I2C_Driver.cpp:(.text+0x114): undefined reference to `bcm2835_i2c_write_read_rs'
/usr/bin/ld: libMLX90640_API.a(MLX90640_RPI_I2C_Driver.o): in function `MLX90640_I2CWrite(unsigned char, unsigned short, unsigned short)':
MLX90640_RPI_I2C_Driver.cpp:(.text+0x258): undefined reference to `bcm2835_i2c_write'
collect2: error: ld returned 1 exit status
make: *** [Makefile:49: examples/test] Error 1
```

```
root@pi4touch:~/mlx90640-library/examples# grep i2c /boot/config.txt 
dtparam=i2c_arm=on
#dtparam=i2c_arm=on,i2c_arm_baudrate=400000
```

```
root@pi4touch:~/mlx90640-library/examples# ./hotspot 
The framebuffer device was opened successfully.
Original 800x480, 32bpp
cnt > 4 error 
```

```
root@pi4touch:~/mlx90640-library/examples# ./test 
Starting...
Configured...
EE Dumped...
frameData timeout error waiting for dataReady 
Segmentation fault
```

```
oot@pi4touch:~/mlx90640-library/examples# ./video 
The framebuffer device was opened successfully.
Original 800x480, 32bpp
Setting output format...
Setting filename...
Add video streamt...
Opening codec...
Allocating image...
Writing header...
[apng @ 0x1f3cb00] Using AVStream.codec to pass codec parameters to muxers is deprecated, use AVStream.codecpar instead.
Encode started..
frameData timeout error waiting for dataReady 
frameData timeout error waiting for dataReady 
frameData timeout error waiting for dataReady 
```

Gadgetoid, 2020-03-08:
Has the sensor worked previously, and where did you buy it from?

albflo, 2020-04-02:
I experience the same issue and on a RPi0 the camera and code works. Did you solve it somehow? One thing could be the cpu clock which gets changed on demand, but I already fixed that but it still occurs.

srscotty, 2020-05-28:
I am finding the same issue when my fps is set too high

closed 2023-01-20 by Gadgetoid