pimoroni/mlx90640-library #48: Error in python experimental library: undefined symbol:  bcm2835_i2c_setSlaveAddress

nobinobooki, 2020-03-08:
I get this error when running test.py in in python experimental library on Raspberry Pi zero

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/MLX90640-0.0.2-py3.7-linux-armv6l.egg/MLX90640.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 670, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 583, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1043, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: /usr/local/lib/libMLX90640_API.so: undefined symbol: bcm2835_i2c_setSlaveAddress

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test.py", line 6, in <module>
    import MLX90640 as x
  File "/usr/local/lib/python3.7/dist-packages/MLX90640-0.0.2-py3.7-linux-armv6l.egg/MLX90640.py", line 17, in <module>
    _MLX90640 = swig_import_helper()
  File "/usr/local/lib/python3.7/dist-packages/MLX90640-0.0.2-py3.7-linux-armv6l.egg/MLX90640.py", line 16, in swig_import_helper
    return importlib.import_module('_MLX90640')
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: /usr/local/lib/libMLX90640_API.so: undefined symbol: bcm2835_i2c_setSlaveAddress

Gadgetoid, 2020-03-09:
This would imply the BCM2835 library isn't being linked in, which I would have thought would fail with a "library not found" error, rather than linking and failing like this.

Looking at `setup.py` I've got an inkling that the library is being erroneously compiled from source without the `-lbcm2835` flag:

https://github.com/pimoroni/mlx90640-library/blob/6fd7791d6b180d998b79e9b86b60ac79b9a525f1/python/library/setup.py#L53-L62

Possible solution would be to uninstall the library, remove the Python `build` and `dist` dirs, then recompile the base library manually by running `make` and `make install` in the root of this repository. Once this is done I'd assume the Python library will pick up the existing base library, rather than trying to compile it's own.

I'll run some tests to see if I'm on the right track.

Gadgetoid, 2020-03-09:
Okay the Python module build fails outright without the libraries installed, so I'm not sure what's happening here.

Could you provide a step-by-step that leads to your issue?

closed 2020-03-11 by nobinobooki