pimoroni/mlx90640-library #50: editing and compiling code

srscotty, 2020-04-13:
Thanks in advance.
I just installed the mlx90640 library and it operates fine with the programs provided in the examples directory.
Now to tweak those programs...
I would like to change the hotspot program three ways
1. change the fps to 32
2. change the size of the display window
3. display the temperature in Farenheit instead of Celsius.

When I use Geany to open the sourefile hotspot.cpp, I can make the edits, but It will not build.
I first saw that I was having problems with #include mlx90640_api.h and #include fb.h
Q1: How do I point to these libraries properly

Secondly, even making no changes at all, compilation fails.
I get the following error
/usr/bin/ld: /tmp/ccDaAH3z.o: in function 'put _pixel_scaled(int, int, int, int, int)'"
hotspot.cpp (.text+0x94): undefined reference in 'fb_put_pixel'
Q2: how do I remedy this error

I get a series of errors ...:undefined reference to 'MLX90640_SetSubPageRepeat.... et al
Q3: how do I remedy these errors

Again, thanks for helping out a noob

Gadgetoid, 2020-04-14:
You should use `make` and the supplied Makefile to build the examples- if you compile just the file on its own, then it'll be missing all of the other code/libraries that it depends upon.

srscotty, 2020-04-14:
I have done that. How do I go about editing one of those example files?

Gadgetoid, 2020-04-14:
If you edit an example and run `make` again, it should pick up the changes and recompile the relevant code.

srscotty, 2020-04-14:
Thank you. That fixed this problem!

srscotty, 2020-04-14:
for Q3: In HOTSPOT,  in the main routine, change put_number(0, 33, hotspot) to put_number(0, 33, hotspot*9/5+32)
That will make the display in Farenheit.

closed 2020-04-14 by srscotty