pimoroni/mlx90640-library #51: landscape

srscotty, 2020-05-24:
Having a great time experimenting with the MLX90640
I want to change the format from a portrait mode to a landscape mode. When I exchange the x and y values in the put_pixel routine, it turns the image landscape, but it is upside down. and the numbers appear backward.
Any ideas on how to turn the image landscape (right side up) and keep the digits under the image?

Gadgetoid, 2020-05-25:
Which example was this? I'm sure I had portrait/landscape code somewhere.

You would potentially have to flip the image vertically, after rotating, which you could accomplish by subtracting whatever the new Y dimension is from the Y height.

IE; `Y = Normal` and `HEIGHT - Y = Inverted Y`

srscotty, 2020-05-25:
I was working with hotspot. 
I kind of got it by switching y and x values, but I would like the ability to rotate 90, 180, 270° etc

srscotty, 2020-06-03:
That’s a cool idea. Thanks.

closed 2023-01-20 by Gadgetoid