pimoroni/mlx90640-library #53: Examples not working

ColdThor, 2021-06-14:
I am pretty new to Raspberry Pi and Raspbian, but I followed the instructions in the readme but whenever I try one of the examples I get "command not found". When I navigate to the examples folder, no such scripts are available there, only some folders.

I am using Raspberry Pi 4, the thermal camera is detected when I use i2cdetect.

Gadgetoid, 2021-06-17:
Have you run the "make" command to build an example?

ie:

```
make examples/fbuf
sudo examples/fbuf
```

Does it give any useful output, or throw any errors when compiling?

ColdThor, 2021-06-18:
Hi,

`sudo examples/fbuf `

is exactly the line returning "command not found"

`make examples/fbuf`

is returning: "make: Nothing to be done for `all'."

I am using the terminal in Raspberry Pi 4, the other commands up to the `make examples/xxx` were seemingly working without errors.

closed 2023-01-20 by Gadgetoid