pimoroni/mlx90640-library #57: Raspberry Pi Zero 2 `make I2C_MODE=RPI` Compilation Error: Undefined Reference

mFarouki, 2022-01-15:
Hello! I am on a new pi zero 2, and am trying a fresh install of this library. I am getting the following error when I run `make I2C_MODE=RPI`:

```
g++ -fPIC -I headers -shared    -c -o functions/MLX90640_API.o functions/MLX90640_API.cpp
functions/MLX90640_API.cpp: In function ‘int MLX90640_GetData(uint8_t, uint16_t*)’:
functions/MLX90640_API.cpp:86:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
g++ -fPIC -I headers -shared    -c -o functions/MLX90640_RPI_I2C_Driver.o functions/MLX90640_RPI_I2C_Driver.cpp
ar rcs libMLX90640_API.a functions/MLX90640_API.o functions/MLX90640_RPI_I2C_Driver.o
ranlib libMLX90640_API.a
g++ -fPIC -shared functions/MLX90640_API.o functions/MLX90640_RPI_I2C_Driver.o -o libMLX90640_API.so
g++ -I. -std=c++11 -std=c++11   -c -o examples/src/test.o examples/src/test.cpp
g++ -L/home/pi/mlx90640-library examples/src/test.o libMLX90640_API.a -o examples/test
/usr/bin/ld: libMLX90640_API.a(MLX90640_RPI_I2C_Driver.o): in function `MLX90640_I2CRead(unsigned char, unsigned short, unsigned short, unsigned short*)':
MLX90640_RPI_I2C_Driver.cpp:(.text+0x84): undefined reference to `bcm2835_init'
/usr/bin/ld: MLX90640_RPI_I2C_Driver.cpp:(.text+0x88): undefined reference to `bcm2835_i2c_begin'
/usr/bin/ld: MLX90640_RPI_I2C_Driver.cpp:(.text+0x90): undefined reference to `bcm2835_i2c_set_baudrate'
/usr/bin/ld: MLX90640_RPI_I2C_Driver.cpp:(.text+0xe0): undefined reference to `bcm2835_i2c_setSlaveAddress'
/usr/bin/ld: MLX90640_RPI_I2C_Driver.cpp:(.text+0x114): undefined reference to `bcm2835_i2c_write_read_rs'
/usr/bin/ld: libMLX90640_API.a(MLX90640_RPI_I2C_Driver.o): in function `MLX90640_I2CWrite(unsigned char, unsigned short, unsigned short)':
MLX90640_RPI_I2C_Driver.cpp:(.text+0x258): undefined reference to `bcm2835_i2c_write'
collect2: error: ld returned 1 exit status
make: *** [Makefile:49: examples/test] Error 1
```

Note: I am able to run `make I2C_MODE=LINUX` with no errors (though quite a lot of deprecation warnings!) Many thanks in advance for your time.

## Steps Taken
Following your README, I have done the following:
- `sudo apt-get update`
- `sudo apt-get install`
   - `libi2c-dev`
   - `libavutil-dev`
   - `libavcodec-dev`
   - `libavformat-dev`
   - `libsdl2-dev`
- Edit `/boot/config.txt` to include `dtparam=i2c1_baudrate=400000`
- `make` steps
   - `make clean`
   - `make bcm2835` 
   - `make I2C_MODE=RPI`

## System Information
- Raspberry Pi Zero 2
- `Linux raspberrypi 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux`
- `PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"`.

Gadgetoid, 2022-01-19:
I don't know if `I2C_MODE=RPI` still works, or ever worked, but I've pushed up a patch to fix the missing `-lbcm2835` that it needed.

mFarouki, 2022-01-19:
Amazing - thanks very much! 🥇

closed 2022-01-19 by mFarouki