pimoroni/mlx90640-library #59: Segmentation fault error with Pi4

cvrsor, 2022-07-18:
Hello, 

I'm having trouble getting the Pimoroni MLX90640 sensor to work with a Raspberry Pi 4 (4GB RAM). Whenever I run any of the examples, I'm getting the following error:
```
bcm2835_init: gpio mmap failed: Operation not permitted
Segmentation fault
```

I've connected the module to the Raspberry Pi's GPIO pins 1, 3, 5, 7, 9.

I've followed the instructions and compiled the bcm2835 library (v. 2.0.22), installed the dependencies, and built the mlx90640-library by running `sudo make all I2C_MODE=RPI`. I've also set `baudrate=400000` by editing the boot/config.txt file.

I'm not sure what else to try, would appreciate any help!

Thanks!

JC53489, 2022-09-24:
I'm having the same issue. Same setup, followed instructions and same error. Any progress on this?

JC53489, 2022-09-25:
Done a little more testing and discovered that I had failed to enable the I2C interface, which can be done with "sudo raspi-config" and looking in the interface section. Hope this helps anyone else who made the same silly mistake 😄

closed 2023-01-20 by Gadgetoid