pimoroni/mlx90640-library #60: Raspberry pi 3 can't go above 4hz

Yours3lf, 2022-08-21:
Hi,

I'm trying to run test.cpp at frame rates above 4hz, but it just gives me the "cnt > 4" error.
4hz does work though.
I'm using a raspberry pi 3 B+ with 1Mhz baudrate
the camera module is the pimoroni MLX90640 breakout from pihut (PIM365)
i2cdetect gives me 0x33
it is the only thing connected to the RPi.
I basically wired up
pi -> mlx90640
3v3 to 3.3v-6v
sda to sda
scl to scl
gnd to gnd

How can you run this at the advertised 64hz?

thanks!

Gadgetoid, 2022-08-22:
High-frequency i2c can be really sensitive to wiring, do you have a picture of your setup?

Yours3lf, 2022-08-22:
[image]

Yours3lf, 2022-08-22:
[image]
[image]

Yours3lf, 2022-08-22:
[image]

Yours3lf, 2022-08-27:
Okay I'll try that

Yours3lf, 2022-09-02:
[image]
[image]
[image]

Yours3lf, 2022-09-02:
hey, tried it today, but the above setup has the same behaviour unfortunately.

Yours3lf, 2022-09-19:
any other suggestion please?

myslide, 2022-09-25:
I build it according the instructions on a Rpi 400. I changed the config.txt. without a reboot and got these error too in ./fbuf. Rebooted, make clean, make I2C_MODE ->./fbuf confirmed the success by 1440x900 but none picture-> Ok, resolution is to hight. The ./test shows the raw pixels with a slow framerate. But as I tried  sdlsclae it satisfied me with the acceptable interpolation and framerate. So these are my experiences. I wired it up with 10cm reused, preconfigured plug and soldered it on the breakout board. Make sure your system is up to date, the required libs are installed (I got warnings with the libavutils reg. the decoder), make clean, make all... . Keep tought!

closed 2023-01-20 by Gadgetoid