pimoroni/mlx90640-library #68: Compilation errors with video.cpp, something with codecs

swekley, 2024-03-01:
I'm trying to compile the library on a pi 4b but It's giving errors with examples/src/video.cpp 

I installed all the dependencies and ran make clean first, the errors happen with both make I2C_MODE=LINUX & RPI

> g++ -DDEBUG -g -I. -std=c++11 -std=c++11 -Wall -Werror -Wno-format-extra-args -Wno-deprecated-declarations   -c -o examples/src/video.o examples/src/video.cpp
examples/src/video.cpp: In function 'void video_encode_start(const char*, int, AVCodecID)':
examples/src/video.cpp:68:5: error: 'av_register_all' was not declared in this scope
   68 |     av_register_all();
      |     ^~~~~~~~~~~~~~~
examples/src/video.cpp:71:37: error: invalid conversion from 'const AVOutputFormat*' to 'AVOutputFormat*' [-fpermissive]
   71 |     avOutputFormat = av_guess_format(NULL, outputfile, NULL);
      |                      ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
      |                                     |
      |                                     const AVOutputFormat*
examples/src/video.cpp:74:41: error: invalid conversion from 'const AVOutputFormat*' to 'AVOutputFormat*' [-fpermissive]
   74 |         avOutputFormat = av_guess_format("apng", NULL, NULL);
      |                          ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
      |                                         |
      |                                         const AVOutputFormat*
examples/src/video.cpp:97:31: error: 'AVFormatContext' {aka 'struct AVFormatContext'} has no member named 'filename'
   97 |     snprintf(avFormatContext->filename, sizeof(avFormatContext->filename), "%s", outputfile);
      |                               ^~~~~~~~
examples/src/video.cpp:97:65: error: 'AVFormatContext' {aka 'struct AVFormatContext'} has no member named 'filename'
   97 |     snprintf(avFormatContext->filename, sizeof(avFormatContext->filename), "%s", outputfile);
      |                                                                 ^~~~~~~~
examples/src/video.cpp:101:33: error: invalid conversion from 'const AVCodec*' to 'AVCodec*' [-fpermissive]
  101 |     codec = avcodec_find_encoder(avOutputFormat->video_codec);
      |             ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                 |
      |                                 const AVCodec*
examples/src/video.cpp:117:25: error: 'AVStream' {aka 'struct AVStream'} has no member named 'codec'
  117 |     if (video_avstream->codec == NULL) {
      |                         ^~~~~
examples/src/video.cpp:123:25: error: 'AVStream' {aka 'struct AVStream'} has no member named 'codec'
  123 |     c = video_avstream->codec;
      |                         ^~~~~
examples/src/video.cpp: In function 'void video_encode_frame(int)':
examples/src/video.cpp:212:11: error: 'avcodec_encode_video2' was not declared in this scope; did you mean 'avcodec_encode_subtitle'?
  212 |     ret = avcodec_encode_video2(c, &pkt, frame, &got_output);
      |           ^~~~~~~~~~~~~~~~~~~~~
      |           avcodec_encode_subtitle
examples/src/video.cpp:226:17: error: 'av_free_packet' was not declared in this scope; did you mean 'av_new_packet'?
  226 |                 av_free_packet(&pkt);
      |                 ^~~~~~~~~~~~~~
      |                 av_new_packet
examples/src/video.cpp: In function 'void video_encode_end()':
examples/src/video.cpp:247:48: error: 'AVStream' {aka 'struct AVStream'} has no member named 'codec'
  247 |         av_freep(&avFormatContext->streams[i]->codec);
      |                                                ^~~~~
make: *** [<builtin>: examples/src/video.o] Error 1

mikerr, 2024-03-09:
This is a problem with libav  / ffmpeg versions above 4.0 ...

swekley, 2024-03-09:
> This is a problem with libav  / ffmpeg versions above 4.0 ...

Ah yeah thats what I was thinking, compiled a version in the 3.x's when I apparentally mixed up vcc and gnd and smelled parts of my sensor burning out 😩 i ended up just giving up completely, but thanks for the feedback though.. if I ever buy a new one at least I know how to continue 😅

mikerr, 2024-03-10:
Fixed with this PR 
https://github.com/pimoroni/mlx90640-library/pull/69

swekley, 2024-03-13:
> Fixed with this PR #69

Thanks!

closed 2024-03-13 by swekley